onelife/Arduino_RT-Thread #5: would you like to support stm32 in platformio under vscode ide?

KiraVerSace, 2020-12-07:
Thank you for your contribution!
HaHa, Are you Chinese?
I develop my project with stm32duino under platformio, it is very conveniently to build a project.
When I use it in a big project, I have to use RTOS, but there is only one choice, stm32duino -freertos, there are many bugs in it, in the past when I use keil I always use rtthread, but now I use the mac, So I have no idea.
would you like to hele us to adapt rtthread to stm32 in platformio, or you can write it down about you how to adapt to samd21, then I can copy your idea, do it by myself.
Thank you!

onelife, 2020-12-10:
Hi @KiraVerSace,
Yes, you are right.
Someone also requested for STM32duino support before. It should not be difficult. However, I don't have such a board currently.
As the STM32duino is popular, I decided to buy a board soon.
At the meantime, you may check the code under flag "ARDUINO_ARCH_SAMD" for SAMD architecture.

onelife, 2021-03-01:
3fb070cf94c4ad909f26ebe6791b41e4a7578d23 provides basic support for STM32 architecture (Nucleo-F767ZI board).

closed 2022-01-02 by onelife